OrcaSlicer/OrcaSlicer #1027: Fully Portable

ValSimDim, 2023-05-10:
First of all, great job with this slicer, best slicer I ever used and by much!

Apologies if this has been already discussed and answered. I really like that the installation of the slicer is just unzip in folder. My problem is that I use the slicer on 4 different win 11 computers and all custom profiles and settings I'm working on are not saved in the installation folder, but in C:\Users\xxxxx\AppData\Roaming\OrcaSlicer folder on the local machine. I'm using a cloud folder to be able to use the slicer from all different computers with just one installation, and that works fine, except I don't have any way of synchronizing the profiles between the different PC's. You can imagine how inconvenient would be to copy from one machine to another a few times a day....

 Is there a way to save everything in the installation folder instead of local.machine :\Users\xxxxx\AppData\Roaming\OrcaSlicer? Any way to modify a config file or something??

kylek29, 2023-05-10:
For what it's worth, I use the -datadir parameter to point a folder within an OrcaSlicer directory and keep mine in a Cloud Drive, it runs just fine this way.

My data structure is like this, to make updating easier:

[Example]
- C:\PortableApps\OrcaSlicer\   [Main Folder]
- C:\PortableApps\OrcaSlicer\Profiles    [Profile Folder]
- C:\PortableApps\OrcaSlicer\OrcaInstall     [Install Directory, this allows me to keep the install separate and swap out versions if need-be]

My Shortcut looks like this:

```
Target: C:\PortableApps\OrcaSlicer\OrcaInstall\orca-slicer.exe -datadir "..\Profiles"

Start In: C:\PortableApps\OrcaSlicer\OrcaInstall 

```

The use of the "..\" on the profile tells the software to move up 1 directory (into the OrcaSlicer directory) and then look for "Profiles".

SoftFever, 2023-05-10:
Thanks @kylek29 for sharing this.
I'm also thinking about to having a setting to specify the location of the user folder(where the profile exists) only

ValSimDim, 2023-05-12:
Thank you so much @kylek29 and @SoftFever ! Works perfectly on all my PC's.
Having a setting for user folder would be absolutely fantastic @SoftFever !

luckyzor, 2023-05-24:
> Thank you so much @kylek29 and @SoftFever ! Works perfectly on all my PC's. Having a setting for user folder would be absolutely fantastic @SoftFever !

Hello, 
can you explain a little more indeph  how you do this?
thank you in advance.

YanceyA, 2023-05-25:
kylek29 Recommendation worked for me great for machine portability with a onedrive setup.

jeronbernal, 2023-06-19:
Ive noticed there is no "profiles" in my appdata/roaming 

i do see a "user" folder but when trying to set my target to "users" or even the orcaslicer folder copied to dropbox/onedrive it isn't opening orcaslicer as usual

Here's my original Superslicer shortcut and my Orcaslicer (attempted) shortcut 

**-SuperSlicer**

**Target** : C:\Users\jeron\Downloads\SuperSlicer_2.5.59.2_win64_230304\SuperSlicer_2.5.59.2_win64_230304\superslicer.exe --datadir "C:\Users\jeron\Dropbox\Apps\SuperSlicer Data\SuperSlicer Profiles"

**Start in** : C:\Users\jeron\Downloads\SuperSlicer_2.5.59.2_win64_230304\SuperSlicer_2.5.59.2_win64_230304

**Attempted but failed OrcaSlicer shortcut :** 

**Target** : C:\Users\jeron\Downloads\OrcaSlicer_V1.6.3_Win64 (1)\orca-slicer.exe -datadir "C:\Users\jeron\Dropbox\Apps\OrcaSlicer\user"

**Start in** : "C:\Users\jeron\Downloads\OrcaSlicer_V1.6.3_Win64 (1)"

i copied my appdata/roaming/orcaslicer folder to my dropbox folder under dropbox/apps/orcaslicer 

my orcaslicer.exe i have stored locally in my "C:\Users\jeron\Downloads\OrcaSlicer_V1.6.3_Win64 (1)

Was hoping anyone could help me with this situation, i'm a bit stumped, i tried --datadir aswell as -datadir

closed 2023-11-21 by github-actions[bot]

AJolly, 2024-03-30:
I wish you could have orca look for a file, and if so have it run in portable mode, rather than forcing a cli param.

PR #6780 (2024-09-15): Check and use data_dir folder if exists under application path
Orca now check if a data directory (`data_dir`) exists at the application path. If so, use it as the data location. Otherwise, fall back to the `--datadir` option specified on the command line.

Fixes #1027