OrcaSlicer/OrcaSlicer #1031: Slice gap closing radius & Classic wall generator V1.6.3

TechDude23, 2023-05-10:
Slice gap closing radius with classic wall generator bug and stl file is ok. 

[image]

[stl file](https://files.printables.com/media/prints/35766/stls/353764_1894b1e6-c546-44a7-b07b-ce674effe081/tier.stl)

Window 10
Version 1.6.2 has the problem with more bars 
Version 1.6.3 is better but still needs 0.001 to fix classic wall bars

SoftFever, 2023-05-10:
I can't repro the issue with default setting.
It would be easier to save and share the project in the future. The project file contains both the model and the settings.
[image]

TechDude23, 2023-05-11:
[Link](https://forum.bambulab.com/t/gap-infill-anomaly/12742)

This might help more

closed 2023-11-21 by github-actions[bot]