OrcaSlicer/OrcaSlicer #1032: Build of AUR pkg fails on EndeavourOS(archlinux based)

goombah88, 2023-05-10:
**Describe the bug**
I have tried twice to build and install the AUR package of orca-slicer 1.6.2-2. It keeps failing on the same part. I would love to try out orcaslicer but I can't get it to build and I can't find a prebuilt AppImage anywhere.

**Desktop (please complete the following information):**
 - OS: EndeavourOS(archlinux based)
 - Version: all latest updates applied

**Build output:**
[ 47%] Building CXX object CMakeFiles/tbb_static.dir/src/tbb/mutex.cpp.o
In file included from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/old/task_v2.cpp:29:
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:263:20: error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]
  263 |         tbb::task& task() {return *reinterpret_cast<tbb::task*>(this+1);}
      |                    ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:222:9: note: used here to mean ‘class tbb::task’
  222 |         task* next_offloaded;
      |         ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:33:7: note: declared here
   33 | class task;
      |       ^~~~
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:118: CMakeFiles/tbb_static.dir/src/old/task_v2.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
In file included from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/scheduler_common.h:43,
                 from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/scheduler.h:24,
                 from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/governor.cpp:25:
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:263:20: error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]
  263 |         tbb::task& task() {return *reinterpret_cast<tbb::task*>(this+1);}
      |                    ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:222:9: note: used here to mean ‘class tbb::task’
  222 |         task* next_offloaded;
      |         ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:33:7: note: declared here
   33 | class task;
      |       ^~~~
In file included from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/scheduler_common.h:43,
                 from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/scheduler.h:24,
                 from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/arena.cpp:23:
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:263:20: error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]
  263 |         tbb::task& task() {return *reinterpret_cast<tbb::task*>(this+1);}
      |                    ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:222:9: note: used here to mean ‘class tbb::task’
  222 |         task* next_offloaded;
      |         ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:33:7: note: declared here
   33 | class task;
      |       ^~~~
In file included from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/scheduler_common.h:43,
                 from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/market.h:26,
                 from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/market.cpp:24:
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:263:20: error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]
  263 |         tbb::task& task() {return *reinterpret_cast<tbb::task*>(this+1);}
      |                    ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:222:9: note: used here to mean ‘class tbb::task’
  222 |         task* next_offloaded;
      |         ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:33:7: note: declared here
   33 | class task;
      |       ^~~~
In file included from /var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/old/concurrent_vector_v2.cpp:24:
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:263:20: error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]
  263 |         tbb::task& task() {return *reinterpret_cast<tbb::task*>(this+1);}
      |                    ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:222:9: note: used here to mean ‘class tbb::task’
  222 |         task* next_offloaded;
      |         ^~~~
/var/tmp/pamac-build-goombah/orca-slicer/src/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:33:7: note: declared here
   33 | class task;
      |       ^~~~
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:90: CMakeFiles/tbb_static.dir/src/old/concurrent_vector_v2.cpp.o] Error 1
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:258: CMakeFiles/tbb_static.dir/src/tbb/governor.cpp.o] Error 1
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:286: CMakeFiles/tbb_static.dir/src/tbb/market.cpp.o] Error 1
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:132: CMakeFiles/tbb_static.dir/src/tbb/arena.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:114: CMakeFiles/tbb_static.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/dep_TBB.dir/build.make:86: dep_TBB-prefix/src/dep_TBB-stamp/dep_TBB-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:345: CMakeFiles/dep_TBB.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...

TexZeTech, 2023-05-25:
@goombah88 I'm assuming you are using https://aur.archlinux.org/packages/orca-slicer either way as far as I know this is not something maintained by the people on here. 

You can install the appimage by installing AppImageLauncher and requirements. You might also want to poke the https://forum.endeavouros.com/ for help or any of the other communities listed https://endeavouros.com/community/ I think you will also need to install webkit2gtk if you don't have it in EndevorOS, I would recommend running the appimage in terminal to see what it complains about.

c2h5oh, 2023-05-31:
This is caused by Arch switching to GCC 13.x and it also breaks BambuStudio. Latest Fedora is also affected. I submitted a set of patches to fix it there ( https://github.com/bambulab/BambuStudio/pull/1882 )  and I'll be opening a PR here too.

c2h5oh, 2023-06-16:
Aur is now fixed: patches are still manually applied, but with the next release even that will no longer be required as PR has been merged.

closed 2023-11-21 by github-actions[bot]