OrcaSlicer/OrcaSlicer #1034: Only 1 wall on top surface won't work properly

rfnovo, 2023-05-11:
Using Orca Slicer 1.62, with the "Only one wall on top surfaces" enabled, but as you can see in this simple geometry, Orca slicer will only apply this setting in the utmost top layer:  

[image]

TechDude23, 2023-05-11:
Looks like they have fixed it with 1.6.3 you can download the beta version or try a modifier might work on the older V 1.6.2 

[image]

liftbag, 2023-05-11:
> Looks like they might have fixed it with 1.6.3 you can download the beta version or try a modifier might work on the older V 1.6.2
> 
> Is this on the Bambu Lab X1 ?
> 
> [image]

Maybe you are on classic wall generator? The issue is related to the arachne wall generator.
This is Orca Slicer 1.6.3-beta.

[image]

liftbag, 2023-05-11:
One of the most duplicate issues at the moment  #379 #459 #547 #582 #709 #763

closed 2023-11-21 by github-actions[bot]