OrcaSlicer/OrcaSlicer #1042: Overhang and Bridge Cooling

NexGen-3D-Printing, 2023-05-11:
**Describe the bug**
When I enable "Force Cooling For Overhangs and Bridges" The fan will come on as expected, then stay on for the entire layer, and will not turn off until the next layer, this for high temp filaments sensitive to part cooling injects a weak layer in the model cause very low layer adhesion.

**3mf File for This Bug**
Will upload a file with issue later.

**To Reproduce**
Force Cooling For Overhangs and Bridges

**Expected behavior**
Expected result would be the same as Cura, fan on for the bridge or overhang, then off for the rest of the layer.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
Orca Arena X1 0.4

**Desktop (please complete the following information):**
 - OS: Winblows 11
 - 1.6.2

NexGen-3D-Printing, 2023-05-24:
Okay, now this bug has extended to the support interface cooling in beta version 1.6.3

I had to disable that as once the fan comes on for the support interface, it stays on for the entire layer of that model, then when the print head retracts and goes to another model, the fan shuts off, does this in RepRap and Klipper, compromises the part when printing sensitive materials, so I have chose to disable the fan features completely now, it just does not work the same as Cura does.

kvnper, 2023-05-24:
I've encountered this too, it destroys some prints due to too much cooling

NexGen-3D-Printing, 2023-05-24:
> 

Can confirm there are others experiencing this issue, I think most people don't notice as most print PLA or PETG, but it really effects ABS, PA and PC based filaments, its a real shame, as everything else in this slicer is really good, very impressed with it.

asychev, 2023-07-21:
I can confirm that, ridiculously annoying bug which REALLY ruins some ABS prints
[image]
Quite often I simply can't use Orca just because of that

asychev, 2023-07-24:
@SoftFever any chance this can get your attention?

SoftFever, 2023-07-25:
@asychev could you attch the project file?

SoftFever, 2023-07-25:
@asychev 
Looks okay on my machine
You have the fan speedup value set to 1 second:
[image]
Turn if off we will get this:
[image]

asychev, 2023-07-25:
@SoftFever This setting cause the issue 
[image]

SoftFever, 2023-07-25:
I tested the project file you shared.
It has the force cooling enabled as I can see.
[image]

asychev, 2023-07-25:
Maybe you are right, I indeed have fan speed up setting enabled and set to 1s, forgot about it.
@NexGen-3D-Printing maybe you have the same issue?

SoftFever, 2023-07-25:
Please always attach project file.
It's really hard and time consuming to guess from screenshots when there could be multple factors affect the result

asychev, 2023-07-25:
[1042-2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12162248/1042-2.3mf.zip)
Layers 67 and 70, layer time looks similar, but fan speed is different

SoftFever, 2023-07-25:
@asychev 
This is how auto cooling/fan works:
The shorter the layer time, the higher the fan speed it needs; and vice versa.
If you switch to the layer time view, you will notice that those layers have longer layer times compared to other layers.

Reference: https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.4.1

SoftFever, 2023-07-25:
I recommend constant fan speed for cooling-sensitive filaments like ABS

asychev, 2023-07-25:
@SoftFever Okay, seems like I got it finally. Thanks for you time, much appreciated

closed 2023-11-21 by github-actions[bot]