OrcaSlicer/OrcaSlicer #1043: Bridge density setting affects anchoring of bridges

mjonuschat, 2023-05-12:
Setting a bridge density smaller than 100% causes bridges to be extruded mid-air / not being anchored to walls.

This is visible in the  preview as well (50% and 25%):
[image]
[image]

How to reproduce:
Reduce bridge density to less than 100%, slice object with bridges, check preview

Version: OrcaSlicer 1.6.2 (Release) on AppleSilicon

mjonuschat, 2023-05-12:
Project file that reproduces the issue:

[Xol 2 - Faceplate.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11458585/Xol.2.-.Faceplate.3mf.zip)

SoftFever, 2023-05-22:
@mjonuschat 
Thanks for reporting this.
Will fix it in the 1.6.3 official release.

mjonuschat, 2023-05-23:
Will do, the preview is looking good, I'll send a test-job later today to verify.

mjonuschat, 2023-05-23:
Prints come out nicely with the nightly build, thanks for the quick fix!

closed 2023-05-23 by mjonuschat