OrcaSlicer/OrcaSlicer #1046: With multiple Filaments, the first one also triggers "filament change"

tommy536, 2023-05-12:
**Describe the bug**
If several filaments are used in a print, a "filament change" is also triggered for the first filament. This makes no sense, since the first filament is already inserted at the beginning of a print.

**Expected behavior**
The Print should start normal without triggering a filament change before the first filament.

**Printer model**
Voron 2.4

**Desktop (please complete the following information):**
 - OS: Windows 10

hliebscher, 2023-05-13:
@tommy536  it make sense for IDEX Printer   after purge line select the right filament !

Mario-vminion, 2023-05-17:
you could make it an option to choose ... if you have idex you can select ... more people don't have idex ...

awabom, 2023-11-05:
If this change is made, IDEX printers will be even harder to use in OrcaSlicer.

closed 2024-02-19 by github-actions[bot]