OrcaSlicer/OrcaSlicer #1047: On Print start (bambu cloud), slicer does not automatically go to device tab.

psiberfunk, 2023-05-13:
Normally during a print start for the Bambu series printers (X1C in this example), after the popup dialog here the user clicks "send".  Then, after sending the job via the Bambu Cloud, the dialog says going to the device tab in 3...2...1...  and then switches to the device tab after the job is uploaded.
[image]

However, in recent versions of orcaslicer, this has broken.. the job is properly uploaded, the dialog disappears, but the app **does not switch** to the proper tab.  Instead if just goes back to the prepare/preview screen.

It seems like whatever function is supposed to switch tabs is not getting called properly ?  I suspect this has something to do with the differences between the Bambu network plugin and some changes made to Orcaslicer.  This still works as expected on mainline Bambu slicer.

bani6809, 2023-05-13:
i repro this bug as well. never switches to the device tab.

psiberfunk, 2023-05-13:
yea, I agree seems a dupe of #1044

psiberfunk, 2023-05-16:
this is fixed by this commit.

mydevpeeps, 2023-05-31:
confirmed this goes to device tab in 1.6.3.

closed 2023-11-21 by github-actions[bot]