OrcaSlicer/OrcaSlicer #1050: Different speeds on occasional (same) layers

DiegoTheWolf, 2023-05-13:
**Describe the bug**
I have a fairly simple part where i expected to run the same speed for every layer (with some gradient).
But occasionally (and i don't know why) the speed is slowed down for an entire layer.

**3mf File for This Bug**
[Schubladenhalter_h57_w23.zip](https://github.com/SoftFever/OrcaSlicer/files/11469281/Schubladenhalter_h57_w23.zip)

**To Reproduce**
Just loaded the file and sliced it.

**Expected behavior**
Not slowing down on occasional layers or an explanation, why those layers are slowed down, because they are the same.

**Screenshots**
[image]

**Printer model**
FLSUN V400

**Desktop (please complete the following information):**
 - OS: Windows 10 22H2
 - Orcaslicer 1.6.2

hliebscher, 2023-05-13:
i have tested with your model  and no issues

Mac M1  Orca 1.6.3 beta

[image]

DiegoTheWolf, 2023-05-14:
I re-sliced it with different paramentes and it had different and less "faulty" layers, but still.
Need to test the beta then

closed 2023-11-21 by github-actions[bot]