OrcaSlicer/OrcaSlicer #1052: klipper MCU 'mcu' shutdown: Timer too close using Arc Fitting on circular (cylindrical) objects

fmonaca, 2023-05-13:
Printing cylindrical objects leaving Arc Fitting selected as per default (and having Klipper with [gcode_arcs] resolution: 0.1) crashes Klipper with the error 

"klipper MCU 'mcu' shutdown: Timer too close"

This happened with Orcaslicer 1.6.2, Windows 11
Printer is a Voron 2.4 350 with latest Klipper/Mainsail/Moonraker

[Container.zip](https://github.com/SoftFever/OrcaSlicer/files/11469408/Container.zip)

lbibass, 2023-05-21:
decrease the arc resolution. 0.2 or 0.3 should be visually indistinguishable but will massively reduce the workload.

closed 2023-11-21 by github-actions[bot]

loutwice, 2024-05-07:
had the same issue, arc fitting causing MCU TIMER TOO CLOSE..  ... always happen on prime tower 
[klippy (7).log](https://github.com/SoftFever/OrcaSlicer/files/15241103/klippy.7.log)

i couldnt passes first hour of printing, deactiating it cures the problem