OrcaSlicer/OrcaSlicer #1064: Wipe function intermittant failure

apalert, 2023-05-14:
Love what has been done lately with seam gaps. However part of the tool kit for minimal gaps is the wipe and I'm finding it intermittently fails. For example in the shot below where I'm using "Wipe while retracting" and "Retract before wipe" of 5%, I'd say roughly 50% of outer loops don't show a wipe and final retraction.   

[image]
 
Switching from Classic wall to Arachne does improve it to perhaps 70% success. Most other tweaking makes it worse. For example, activating "Precise wall" drops success to about 20% and not having some positive value set for "Z-hop" kills it altogether except for the final wipe in a layer.

What you see in the Preview shot above is reflected in the model. 

[image]

Seams A and B were both wiped and they are pretty good IMHO. Seams C, D and E were not wiped. C and E were not preceded by a wipe either which may explain the appearance of over-extrusion at the start of their loops. Or maybe not, D is lumpy as well :)

When I step through a layer where a wipe is missing, as the loop closes I see no code for the expected wipe/retraction.

[image]

Printer: Bambu Labs X1 Carbon
Desktop: macOS Monterey 12.6

Project file:
[Orca ASA APS51 001X9.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11471873/Orca.ASA.APS51.001X9.3mf.zip)

kellyinvestor, 2023-05-15:
This is a bug caused by "Avoid crossing wall".

bambulab/BambuStudio#965

apalert, 2023-05-15:
Disabling "Avoid crossing wall" does improve it quite a bit to maybe 80% success but doesn't eliminate the problem entirely.

apalert, 2023-05-16:
I take that back. The remaining "failures" were due to my Travel Distance Threshold being set to 2mm. Changed it to 0.1mm and successful wipes are now at 100%! Many thanks for alerting me to the true cause of the bug. I can live without using Avoid Crossing Wall.

killerline10168, 2023-08-26:
I feel like this is still a bug or at least an unintended feature. I personally want my nozzle to wipe every time and everywhere, like how Cura does it. For it to work I need to lower the Travel Distance Threshold, thus forcing it to retract at EVERY nozzle move (even top/bottom infill), to make wiping work. Is there a possibility to make this a separate feature, so that we are able to choose how it wipes?

closed 2023-12-03 by github-actions[bot]