OrcaSlicer/OrcaSlicer #1065: Problem during Tool Change

Durahl, 2023-05-14:
Greetings!

I'm currently trying to setup **_OrcaSlicer 1.6.2_** for use with an ERCF on my Voron V2.4 ( currently using with SuperSlicer ) but am having problems during the initial phase of a Tool Change - [Video of the Issue](https://youtu.be/x2wNMdYcCR4).

**In SS the following Steps are being sent to the Printer / ERCF:**
- Print last line with current Filament
- Raise Tool Head and move above Purge Tower
- Perform Tip Forming Procedure with the Tool Head making use of the Purge Tower
- Extract current Filament
- Change Filament
- Load new Filament
- Purge on Purge Tower
- Return to Print and continue

**In OS the following Steps are missing / Incomplete:**
- _Raise Tool Head and move to Purge Tower_
- Perform Tip Forming Procedure _with the Tool Head making use of the Purge Tower_

As such, the Tool Head will perform the Tip Forming right on top of the so far printed section squirting a healthy dose of Filament onto the part that would normally happen on the Purge Tower effectively ruining the part due to the blob that has formed on it.

Does OS require a setting to be enabled for the Tool Head to not be remaining stationary right on top of the Part before doing the Tip Forming or is OS simply not capable of being used with anything other than the AMS?

kriffo88, 2023-06-03:
Hi, I also encountered the usual problem, did you manage to solve it somehow?

Durahl, 2023-06-03:
Yea... with a crude hack adding a bit of G-Code prior to the Tool Change Command moving the Tool Head off the part into a corner of the Build Plate:
```
G1 X345 Y345 F24000
T[next_extruder]
```
With `X0 Y0` being the front left Corner the above example will be moving it to the almost max rear right.

kriffo88, 2023-06-03:
I tried and it seems to have improved the filament change. Can you share your print start and configuration on Orcaslicer? Thank you very much!

dohlin, 2023-06-20:
Really wish proper single extruder multi-material was an option on OS.  I, too, am coming from SS due to the lack of maintenance but this is a non-started with my main printer and ERCF system.  Bummer.

Holy006, 2023-07-29:
Hi, I'm happy to join too. With a link for enderwire:
Added to change filament g-code (orca slicer)
G1 X217 Y231 F15000
T[next_extruder]

it somehow works but it's not 100%. Are there other commands that would be better?

jonathanalevi, 2023-08-05:
From what I can tell, there's no cohesive guide for setting up OS with ERCF, huh?

closed 2023-11-21 by github-actions[bot]