OrcaSlicer/OrcaSlicer #1067: Infill/wall overlap specifically for bridges

hutje, 2023-05-14:
**Is your feature request related to a problem? Please describe.**
When changing the infill/wall overlap, it also changes the normal infill overlap. But for bridges with a bigger gap, it is necessary to have more overlap. 

**Describe the solution you'd like**
An option to change the overlap for bridges without the need to change all the overlap.

**Additional context**
As you can see I set overlap to 110% to ensure a good anchorpoint. But for the other lines it is too much. (I have set the bridge-flow to 0.7 and 70% density)
[image]

ncr, 2023-08-13:
I'm having the same issue with bridges and while the infill/wall setting helps, it would be very useful to have such setting specifially for the bridges. We have some specific temperature / fan settings that we can't tweak but still need sound bridges and this is about the only setting that helps but on the other hand it messes up the infill a bit.

closed 2023-11-21 by github-actions[bot]

duerk-de, 2024-10-21:
I would like to add a comment since I am also facing this issue and did not find any suitable workaroud, yet.

The first thing I noticed is, that switching off sparse infill changes things (see images):
[image]
[image]
The sparse infill is substituted by a bridge!

Nevertheless on thin walls there is practically no overlap.

Increasing Infillwall overlap does not add any overlap to thick walls:
[image]

IAmTheDigitalCheese, 2026-10-01:
I would also like to see a separate setting for bridge anchor/overlap % that is separate from the infill/wall overlap setting.. This would help quite a bit.