OrcaSlicer/OrcaSlicer #1068: Missing Minimal filament extrusion length skirt option

wight554, 2023-05-14:
Very useful option available in PrusaSlicer but missing here, it's way easier to have constant minimum skirt than setting it per object (e.g for larger objects you need less loops than for smaller ones)
[image]

wight554, 2023-05-15:
Ended up using PRIME_NOZZLE from [here ](https://www.reddit.com/r/VORONDesign/comments/qztck1/comment/hlpoxq3) for now. But this option would be much better

racemaniac, 2023-06-18:
also a +1 for me, really missing this setting

TANKIST73, 2024-02-12:
I use this option, in PrusaSlicer, to clean the nozzle. There is a great lack of this in Orca.

jpbnto, 2024-02-14:
i need this setting too. hopefully it will be added feature.

igiannakas, 2024-05-13:
PR above implements this feature. Please head over to the PR, download the build file from the actions tab and test the feature. Feedback welcome, in the PR above directly please.

jpbnto, 2024-05-13:
> PR above implements this feature. Please head over to the PR, download the build file from the actions tab and test the feature. Feedback welcome, in the PR above directly please.

thank you

Eldenroot, 2024-05-14:
Hi, please could you take a look, thank you 

#3718 

@igiannakas

igiannakas, 2024-05-14:
@Eldenroot saw this when searching for issues. Again the code seems to be there, the option just removed. I may implement this next if there is enough of a need. Personally I haven’t used draft shields with orca having an enclosed Voron + BBL but I can see the rationale. In the meantime you just check how many layers the model is and set your skirt height to the same number of layers.

igiannakas, 2024-05-14:
So... I've implemented it too @Eldenroot

Eldenroot, 2024-05-14:
Thank you. It is not good that some useful features are just hidden. Because variability is a key, we have different printers etc.

What else could be easily enabled and could be useful?

igiannakas, 2024-05-14:
> Thank you. It is not good that some useful features are just hidden. Because variability is a key, we have different printers etc.
> 
> What else could be easily enabled and could be useful?

Not much more from what I see immediately - the majority of available options are enabled. There are a couple related to multi extruder printing but I bet that these are not able to just be switched on as there is backing code that is more likely than not missing.

closed 2024-05-20 by SoftFever