OrcaSlicer/OrcaSlicer #1069: Crash while cancelling calibration

wight554, 2023-05-14:
**Describe the bug**
When trying to prevent creating new calibration project by clicking cancel in popup, application still tries to create new project and crashes immediately 

**3mf File for This Bug**
Any would work

**To Reproduce**

1. Add some changes to current object
2. Click calibration and select Flow Rate -> Pass 1
3. See popup that lets you: 1) save 2) discard 3) cancel, click cancel
4. Calibration models generated, but application crashes

**Expected behavior**
Should not proceed calibration

**Screenshots**

[image]

**Printer model**
Ender 3

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

wight554, 2023-05-14:
Got the log:
```
1:57:52: OrcaSlicer got an unhandled exception: invalid stof argument
1:57:52: OrcaSlicer got an unhandled exception: invalid stof argument
```

hliebscher, 2023-05-16:
at Mac M1  Orca 1.6.3 works and you have only 2 buttons -> save and cancel
the best way for calibration is an empty plate
delete all objects then make calibration...

wight554, 2023-05-16:
> at Mac M1 Orca 1.6.3 works and you have only 2 buttons -> save and cancel the best way for calibration is an empty plate delete all objects then make calibration...

In my case I've accidentally clicked calibration and wanted to cancel it, but instead of cancelling it, got crash and lost all my config changes

closed 2023-11-21 by github-actions[bot]