OrcaSlicer/OrcaSlicer #1070: X-Ray View, and WTH is Orca Arena X1C?

MikeOxlong83, 2023-05-15:
X-Ray is already available on the preview tab but only while slicing is in progress, then it disappears. It would be nice to have a toggle to turn it on and off, it's a very useful feature to have.

Also during setup there's a seperate fleet of BambuLabs printers labelled as Orca Arena X1c/X1/P1P. For those of us who own these printers it would be nice to know what the hell these other ones are, if they do us any good etc.

cereal7802, 2023-05-18:
OrcaArena printers
A built-in profile with identical Bambulab X1C settings but for Klipper printers. (note: don't forget to enable gcode_arcs support in your [printer.cfg](https://github.com/SoftFever/OrcaSlicer#note))

this is a note from the 1.6.2 release notes.

MikeOxlong83, 2023-06-01:
Good to know, whaddabout x-ray doe?

closed 2023-11-21 by github-actions[bot]

AJolly, 2024-02-23:
I'd really love x-ray view.

SeeTurtlee, 2024-12-07:
> I'd really love x-ray view.

Me too! I have a few parts that I have to use a decent amount of modifiers in tight hidden spaces. Having to switch to preview mode back and forth to see it and adjust it is very tedious. Especially for long slices! Please add this feature if possible!!