OrcaSlicer/OrcaSlicer #1071: Height Range Modifiers

jlew715, 2023-05-15:
**Is your feature request related to a problem? Please describe.**
Yes - the existing Adaptive Layer Height feature, event when adjusted manually, will force transitions that are not a multiple of the printer's step increments, leading too poor quality on these transitional layers.

**Describe the solution you'd like**

1. Direct port of BambuStudio's Height Range Modifiers feature.
or
2. Add the Layer Height property to the Modifier Objects.
or
3. In the Adaptive Layer Height menu, add the ability to explicitly specify the height on a per-layer basis.

**Describe alternatives you've considered**
For parts that require layer height to change abruptly (faster than the sharpest curve in Adaptive Layer Height), I have to use BambuStudio as this is not supported in Orca. This leads to other issues (i.e. pressure advance) as certain Orca features are not supported in BambuStudio.

**Additional context**
I see that many folks have asked for this in the past, but perhaps now that option 1.) above is built into BambuStudio, porting it over would be less burdensome? Even if there was a way to specify the minimum Z-step of your printer, and have Adaptive Layer Height use that, it would be a massive improvement. Even at the sharpest Adaptive Layer Height curve, the printer is being asked to do nonsense layer heights such as 0.113255mm when transitioning between (for example) 0.08mm and 0.16mm. This leads to quality degradation in some scenarios because the printer cannot adequately meet the requested layer height.

liftbag, 2023-05-16:
Height range modifier is already present in the latest development versions. If you don't want to use a development version, just wait for the next release 1.6.3. Luckily SoftFever is very active, I don't think you have to wait long.

jlew715, 2023-05-16:
@liftbag that’s great news! Might be a silly question, but do the development versions exist as binaries anywhere?

shauneccles, 2023-05-17:
Yes - every build is saved as part of the Actions workflow.

https://github.com/SoftFever/OrcaSlicer/actions

Find the platform you're after, select the build and the Nightly artifact will be attached to it.

liftbag, 2023-05-17:
> @liftbag that’s great news! Might be a silly question, but do the development versions exist as binaries anywhere?

Sure! Go to actions, choose the latest workflow compatible with your operating system and download the artifact.

closed 2023-11-21 by github-actions[bot]

Felix14-v2, 2024-12-05:
Stupid bot