OrcaSlicer/OrcaSlicer #1073: Filament G-Code not preserved after restart

Sidirius, 2023-05-15:
**Describe the bug**
After setting filament specific G-Code and saving the setting is not preserved after restart of Orca Slicer.

**To Reproduce**
Take a user-preset filament, enter specific g-code in the advanced tab, save and restart orca slicer

**Expected behavior**
Settings being saved and preserved

**Printer model**
Voron 2.4 R2

**Desktop (please complete the following information):**
 - OS: Mac OS 13.3.1 (a)
 - Version Orca Slicer 1.6.2 Mac AppleSilicon

Sidirius, 2023-05-16:
> @Sidirius
> at mac m1 13.3.1 and OrcaSlicer 1.6.3beta it works
> have you saved the project file too ?

When I save the Project its embedded in the Project. That works.
Will try the 1.6.3beta. Probably its already solved. Thx

julie777, 2023-05-21:
There is a preference to prevent syncing of profiles to the cloud.

TonyK54, 2023-05-29:
On Orca 1.6.2. For my Ender 5 Pro (not sure on the FLSun v400, haven't tried it yet) I also lose my custom filament settings. 

I do not have a cloud account. 

This is rather annoying.

closed 2023-11-21 by github-actions[bot]