OrcaSlicer/OrcaSlicer #1076: Fault in the model after opening X-Y hole size compensation

ChiMufeng, 2023-05-15:
This is a new issue that I discovered today. The model has been checked and there are no issues. I wanted to add a small component to this gyroscope, but I did not reserve the size. I wanted to enable X-Y hole size compensation to alleviate the installation pressure. However, when I finished slicing, I found that the model had serious faults. When I restored the X-Y hole size compensation value to 0, this problem disappeared, I would like to inquire if there is a way to enable X-Y hole size compensation without this fault issue
[image]

ChiMufeng, 2023-05-15:
这是模型文件
[666.zip](https://github.com/SoftFever/OrcaSlicer/files/11482580/666.zip)

ChiMufeng, 2023-05-15:
This is the model file
[666.zip](https://github.com/SoftFever/OrcaSlicer/files/11482581/666.zip)

ChiMufeng, 2023-05-15:
[image]

hliebscher, 2023-05-16:
it is the same at Mac M1 Orca 1.6.3beta 
bigger values produce more overhang lines as error in arachne and classic
value 0 then it is OK 

[image]

hliebscher, 2023-05-16:
at this areas the complete layers lost...  
[image]

hliebscher, 2023-05-23:
@SoftFever  can you look at this problem again ?

SoftFever, 2023-05-24:
Looks like it has trouble detecting holes when there are nested holes.

closed 2023-11-21 by github-actions[bot]