OrcaSlicer/OrcaSlicer #1079: Overhangs, bridges and gap fill velocities are not subject to the maximum volume flow limit

FillPavS, 2023-05-16:
Overhangs, bridges and gap fill velocities are not subject to the maximum volume flow limit in the plastic settings. This causes plugging on soft plastics that cannot be printed quickly. This can be changed manually in the speed settings, but it seems to me that the flow limit should limit all speeds.

[image]
[image]

hliebscher, 2023-05-16:
witch os witch slicer ? and witch model file ?

FillPavS, 2023-05-16:
UPD. Sorry, bridges are fine. Only overhangs and gap fill

FBrand, 2023-07-02:
Same Bug, flow of ~30mm³/s instead of 4mm³.
OS: Manjaro Linux 22.1
Orca Slicer 1.6.3 (AppImage)
Model: 3DBenchy

[image]

[image]

hendrikhaehnel, 2023-07-07:
Seems to be the same issue as: https://github.com/SoftFever/OrcaSlicer/issues/852

Workaround: Use "classic mode" for slow down on overhangs. Max. flow is respected in that mode.

closed 2023-11-21 by github-actions[bot]