OrcaSlicer/OrcaSlicer #1080: Can't print anything with orca slicer

grindfreak1981, 2023-05-16:
So, i already posted this issue with a second version of the OrcaSlicer i think, but there was no solution. I can prepare and slice the files, but when i press "print plate", it doesn't do anything. There is no popup window where you select if you want flow calibration, bed leveling, use of ams and timelapse. I'm confused. The older soft fever works fine, original bambu studio works fine. Orca Slicer doesn't. Tried every new version of it, and nothing happens. I already cleaned my pc of all bambu products, deleted the registry files, reinstalled. And same thing keeps happening. Sorry for another thread, but i can't find a solution for this. 

I'm using windows 11, latest version. Keep in mind, that orca slicer used to work on my pc, but then one day it suddenly stopped. Didn't do any update or anything, it just started doing this. Please help

hliebscher, 2023-05-16:
witch printer ?  how is the printer connected? witch version OrcaSlicer

TechDude23, 2023-05-16:
I don't know if this will help but double check your windows firewall is not blocking the software ? 

Then the network plugin will download and install so if you did not see the network plugin get install my guess is your firewall is blocking you.

closed 2023-11-20 by github-actions[bot]