OrcaSlicer/OrcaSlicer #1083: Support expansion & Interface layers = 1.6.3 Beta - Bug or Feature

TechDude23, 2023-05-16:
Normal support expansion works but the interface layers will not expand; they stay with the part size so you get rough edges or we need an option setting to expand interface layers also.

Orca slicer 1.6.3 Beta
Window 10 
Bambu Lab X1

hliebscher, 2023-05-17:
@TechDude23  has the org BambuStudio the same?

TechDude23, 2023-05-18:
> @TechDude23 has the org BambuStudio the same?

Yes Bambu Lab Studio will do it also what I found out is the snug support option will do it but you can not modifier the supports when you need a part that needs multi options of support types so I used my other slicer for now it would be nice if you could override it with a modifier in the future :)

closed 2023-11-20 by github-actions[bot]

rstanislav, 2024-01-12:
Bump.... this bug still present, support expansion only works in normal (not tree) mode and partially with strong tree type. Without expansion edges are rough and most of time a bit curled.

SBachman, 2024-10-29:
I went through the currently open issues and didn't see anything related but did find this. 

I'm seeing the same issue where tree supports ignore interface expansion, aka "Normal support expansion". Tree supports are great but for some prints without this option or using a different material for the interface I have to use normal supports.