OrcaSlicer/OrcaSlicer #1084: Camera stream does not work on Ubuntu

RobRobM, 2023-05-17:
**Describe the bug**
When starting the camera stream on Ubuntu I get an error message that the codecs are missing (see screenshot). All necessary codecs are installed.

**To Reproduce**
Press the play button in the device tab to start the camera feed and the error message appears.

**Expected behavior**
The camera stream should appear.

**Screenshots**
[image]

**Printer model**
Bambu Labs X1C

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 23.04

hliebscher, 2023-05-17:
and with BambuStudio works? have you installed the required codecs?

RobRobM, 2023-05-17:
Yes I have. I even tried reinstalling them.

closed 2023-11-20 by github-actions[bot]