OrcaSlicer/OrcaSlicer #1085: Error opening appImage on debian

scrhall, 2023-05-17:
**Describe the bug**
Download release 1.6.2 on Debian and when executed:
```
% ./OrcaSlicer_ubu64.AppImage                                                                                                                                                                                                                                  ~ scr-desktop
[2023-05-17 14:12:03.788816] [0x00007f15289889c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/scrhall/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/scrhall/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:15526): GLib-GObject-WARNING **: 14:12:04.416: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.416: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15526): GLib-GObject-WARNING **: 14:12:04.416: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.416: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15526): GLib-GObject-WARNING **: 14:12:04.418: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.418: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15526): GLib-GObject-WARNING **: 14:12:04.418: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.418: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15526): GLib-GObject-WARNING **: 14:12:04.423: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.423: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15526): GLib-GObject-WARNING **: 14:12:04.424: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.424: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.470: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.473: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.473: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.474: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.474: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.474: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.474: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.474: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.474: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.485: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.486: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.486: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.661: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.661: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.661: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15526): Gtk-CRITICAL **: 14:12:04.661: gtk_widget_set_size_request: assertion 'width >= -1' failed
orca-slicer: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
[1]    15526 abort      ./OrcaSlicer_ubu64.AppImage

``` 

**To Reproduce**
Download and execute

**Expected behavior**
Open the app

**Desktop (please complete the following information):**
 - OS: DEBIAN
 - Version: 11.7

Wiggen94, 2023-08-24:
Same, tested on 1.6.5, but it was also present in 1.6.4 for me on Arch.

i-hayes, 2023-10-17:
Same error on Opensuse Tumbleweed, version: OrcaSlicer_Linux_V1.7.0.AppImage

charely6, 2023-11-12:
Same on Zorin OS (running on a macbook pro) with a 1.8 release Candidate 2

xantrino, 2024-01-13:
[2024-01-13 10:21:20.773805] [0x00007f47ebbdcd40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/gn0me/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/gn0me/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.136: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.136: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.137: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.137: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.138: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.138: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.139: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.139: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.140: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.140: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.140: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.140: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.141: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.141: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.142: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.142: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.143: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.143: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11532): GLib-GObject-CRITICAL **: 10:21:21.143: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:11532): Gtk-CRITICAL **: 10:21:21.143: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
EGLDisplay Initialization failed: EGL_NOT_INITIALIZED
Cannot create EGL context: invalid display (last error: EGL_SUCCESS)

Unfortunately, the errors are becoming more and more common. :( OrcaSlicer doesn't even start. You have no choice but to use cura or prusaslicer. Unfortunately

greeze

closed 2024-04-20 by github-actions[bot]