OrcaSlicer/OrcaSlicer #1091: X1C Webcam not working if Bambu Studio is not installed

duracell, 2023-05-18:
**Describe the bug**
If OrcaSlicer is installed without Bambu Studio the webcam is not working with a error message that the dll is not registered, everything besides the webcam seems to work.
After Installation of Bambu Studio, everything is working.

**To Reproduce**
Install OrcaSlicer without Bambu Studio.

**Expected behavior**
Webcam should work without a installation of Bambu Studio.

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 22H2

Maybe #811 is relevant for this problem.

nofuturekid, 2023-06-30:
I see the same with Windows 11 (fresh install)

nofuturekid, 2023-06-30:
Found a fix. We are missing a registry entry:

`
HKEY_CLASSES_ROOT\bambu
    Source Filter    REG_SZ    {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}
`

landomando, 2023-06-30:
> Found a fix. We are missing a registry entry:
> 
> ```
> HKEY_CLASSES_ROOT\bambu
>     Source Filter    REG_SZ    {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}
> ```

This worked for me, can confirm the fix.

zimgir124, 2023-07-06:
Can confirm. 
New text file then copy and paste this:
`Windows Registry Editor Version 5.00

[HKEY_CLASSES_ROOT\bambu]
"Source Filter"="{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}"

` save as a .reg file and open.