OrcaSlicer/OrcaSlicer #1095: flow at first layers.

Ziv831, 2023-05-19:
i need to set up a lower first-layer flow. 

when I print with PETG, my 2 first layer need a smaller flor rate' if it is possible.

hliebscher, 2023-05-19:
it is easy ... add a high range modifier than can you change anything... layer hight flow 
go to prepare / select model / right click and ad the modifier....

Ziv831, 2023-05-19:
Ok, that work fine,  but I meant that in the global process, it would be a "first layer flow " modifier 
Just like in Cura slicer,  I find it very useful. 

Ziv

hliebscher, 2023-05-19:
flow first layer can you define under quality

PowerWiesel, 2023-10-22:
> it is easy ... add a high range modifier than can you change anything... layer hight flow go to prepare / select model / right click and ad the modifier....

thx for the workaround - I would have never found this myself :laughing: 

> flow first layer can you define under quality

where? Was the option removed in the meantime?
[image]

> Ok, that work fine, but I meant that in the global process, it would be a "first layer flow " modifier Just like in Cura slicer, I find it very useful.
> 
> Ziv

100% 
While I really like some of the Orca features, I really miss some Cura features - like flow rate settings for every line type and of course first layer.

closed 2024-03-16 by github-actions[bot]

nuxeh, 2025-03-29:
Bottom surface flow appears to only affect the solid infill, and not the perimeters, i can provide screenshots to prove so, but i assure you that it's the case from what i've seen. I would really like to tweak first layer flow from settings, including walls, without having to use a height range modifier. I see word online that there is an "initial layer flow" concept in developer settings, but for the life of me I can't find it, despite having enabled developer settings.

nuxeh, 2025-03-29:
Also, I can't any obvious way to change flow ratio with a height range modifier, either

haski369, 2025-03-30:
Would like this too, it was very useful in Cura

xamindar, 2026-05-29:
I can't believe this is a not planned feature in orca. It is absolutely important to have for PETG. My printer needs a 80% flow rate for proper dimensions and no over extrusion (E steps are perfect already), but the first layer still needs to be a bit higher for good bed adhesion.