OrcaSlicer/OrcaSlicer #1098: Flow rate test pass 2 layer 1 artifacts

HookyQR, 2023-05-20:
**Describe the bug**
Small issue, but obvious.

Extra inner walls drawn on first layer of Flowrate test - pass 2 with first layer < 0.29mm.

Present at first layer hight of 0.289 and below, but not at 0.29

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

[Flowrate Test - Pass2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11521688/Flowrate.Test.-.Pass2.3mf.zip)

**To Reproduce**
Slice per attached 3mf

**Expected behavior**
No extra inner walls

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Printer model**
Tronxy X5SA
 
**Desktop (please complete the following information):**
 - OS: MacOS
 - Version Ventura 13.2 (22D49)
 - Orca version: 1.6.1

hliebscher, 2023-05-20:
At first take the latest Version or Beta 1.6.3

SoftFever, 2023-05-21:
@HookyQR 
OS will automatically set layer height based on the profile.
Did you modify the layer height after the test was generated?

HookyQR, 2023-05-23:
@SoftFever 
Both yes and no. The artefacts are there with the default (0.25 initial layer + 0.2 layer). I then modified and re-sliced to find out when they went away.

Checked with 1.6.3 beta and it's still there.

HookyQR, 2023-05-23:
Note that all 3 artefacts are the same, just transposed. 🤷

HookyQR, 2023-05-23:
I've taken a look at the source file in blender, and there's nothing there that could cause this, so I'm assuming it's a quirk of the slicer. 🤷

closed 2023-11-20 by github-actions[bot]