OrcaSlicer/OrcaSlicer #1107: User process profiles are not loading (1.6.3-beta)

cpasoft, 2023-05-21:
**Describe the bug**
The user process profiles are not loading.

**To Reproduce**
Create a new project. Modify a process and click on the disk to save it. Give it a name. Go to home and create a new project. On the process selector there the new process create on the first step, everything seems right. Close Orca (or open another instance), create a new project, go to the process selector and the user process is gone.

**Expected behavior**
I want the user process back!!

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version OrcaSlicer 1.6.3-Beta
 - Printer: Flsun V400

cpasoft, 2023-05-21:
On the debug file, I can see:
2023-05-21 18:26:40.734589[Thread 0x000014c0]:Preset incompatibla, not loading: prueba

fr3ak2402, 2023-05-21:
@cpasoft

Yes, you have to change something in your user profile (prueba) to make it compatible again.

macOS:
/Users/{username}/Library/Application Support/OrcaSlicer/user/default/process/prueba.json

Windows:
C:\Users\{username}\AppData\Roaming\OrcaSlicer\user\default\process\prueba.json
You have to open this once and then adjust the following line.

"inherits": "0.20mm Standard @FLSun V400",

Then it is compatible again.

cpasoft, 2023-05-21:
I've done it, but still not working...
[image]

fr3ak2402, 2023-05-21:
@cpasoft 

Then you have to rename these files once.

Then create a new user process once, which you name prueba again.

Then this should show up in the folder.

Then open the new file and the old one too. Then copy all fields into the new file. 

Then you should have everything back.

cpasoft, 2023-05-21:
Ok... I've found the error. I had to edit the line:
"version": "0.0.0.0", -> "version": "1.6.0.0",
Now it seems to work, but if you generate a new one again, it still failing... and you must edit the new one again...

fr3ak2402, 2023-05-21:
Perfect. 

I still had this plan, this had worked in any case.

cpasoft, 2023-05-21:
I think the problem comes from saving a process profile from an old 3mf.

fr3ak2402, 2023-05-21:
Yes, because the process has also been changed. It is just worked on the whole processes of all printers

michaelV400, 2023-05-29:
[image]

I compared FLSun.json files from v1.6.2 and v1.6.3 beta and official, this line was missing.

fr3ak2402, 2023-05-29:
@michaelV400 

Yes that is correct, but the file has been renamed from `0.20mm Standard @FLSun` to `0.20mm Standard @FLSun V400`.

The content of the files is the same up to a few lines because of the name change. 

Therefore, the line is no longer needed, you need to adjust everything in your self-made profiles so that wherever `0.20mm Standard @FLSun` is written, it becomes `0.20mm Standard @FLSun V400`.

closed 2023-11-06 by cpasoft