OrcaSlicer/OrcaSlicer #1108: filament profile presets erased

TailSMeDiA, 2023-05-21:
**Describe the bug**
my pre-tuned filament presets have been erased after restarting due to cloud services not sending my print.  they are no longer in the "application support" folder and when i try to sync, they do not show up either.
i did check the orcaslicer application support folder and under user there is a default folder and a folder with numbers but neither have my filament presets

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
[NewSpool_PETG.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11525648/NewSpool_PETG.3mf.zip)
[recent logs from today.zip](https://github.com/SoftFever/OrcaSlicer/files/11525669/recent.logs.from.today.zip)

**To Reproduce**
i am not sure, i tried to send a print using send, and it completed, but was not on the printer.  i then tried to print, but it said the cloud services were down.  
I saved the project that had the filament preset above with "polymaker polylite abs" i tuned myself.
when i reopened orcaslicer, all filament presets were gone but my slicing profile was still there.  i even opened other files i saved with the same filament preset and it would not come back.
i logged in and out, turned on and off sync(it was turned on when this happened) but nothing makes them come back.
i had 4-6 filaments i tuned myself saved and even checking the "application support" folder they are gone.

**Expected behavior**
i never had an issue with my filament profiles disappearing even when orcaslicer has crashed before, or when the cloud service was unavailable

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
bbl x1c

**Desktop (please complete the following information):**
 - OS: mac os
 - Version 13.4

TailSMeDiA, 2023-05-21:
here is a log i found that has my filament profile names:
2023-05-20 21:59:46.248270[Thread 0x00007ff84e2e2640]:Found the Preset locally: 0.25mm Standard @BBL X1C
2023-05-20 21:59:46.248274[Thread 0x00007ff84e2e2640]:load_user_presetpreset 0.25mm Standard @BBL X1C's update_time is eqaul or newer, cloud  update_time 1683595661, local update_time 1684568426
2023-05-20 21:59:46.248302[Thread 0x00007ff84e2e2640]:Found the Preset locally: Amolen Rainbow PetG
2023-05-20 21:59:46.248312[Thread 0x00007ff84e2e2640]:load_user_presetpreset Amolen Rainbow PetG's update_time is eqaul or newer, cloud  update_time 1684474370, local update_time 1684474370
2023-05-20 21:59:46.248331[Thread 0x00007ff84e2e2640]:Found the Preset locally: Amolen Transparent PETG
2023-05-20 21:59:46.248334[Thread 0x00007ff84e2e2640]:load_user_presetpreset Amolen Transparent PETG's update_time is eqaul or newer, cloud  update_time 1684479786, local update_time 1684479786
2023-05-20 21:59:46.248352[Thread 0x00007ff84e2e2640]:Found the Preset locally: Default 0.4 nozzle Bambu Lab X1 Carbon
2023-05-20 21:59:46.248356[Thread 0x00007ff84e2e2640]:load_user_presetpreset Default 0.4 nozzle Bambu Lab X1 Carbon's update_time is eqaul or newer, cloud  update_time 1684100381, local update_time 1684100381
2023-05-20 21:59:46.248371[Thread 0x00007ff84e2e2640]:Found the Preset locally: Generic ABS - Copy
2023-05-20 21:59:46.248374[Thread 0x00007ff84e2e2640]:load_user_presetpreset Generic ABS - Copy's update_time is eqaul or newer, cloud  update_time 1684375874, local update_time 1684375874
2023-05-20 21:59:46.248398[Thread 0x00007ff84e2e2640]:Found the Preset locally: Overture Silk PLA
2023-05-20 21:59:46.248401[Thread 0x00007ff84e2e2640]:load_user_presetpreset Overture Silk PLA's update_time is eqaul or newer, cloud  update_time 1684043280, local update_time 1684043280
2023-05-20 21:59:46.248438[Thread 0x00007ff84e2e2640]:Found the Preset locally: Overture TPU
2023-05-20 21:59:46.248442[Thread 0x00007ff84e2e2640]:load_user_presetpreset Overture TPU's update_time is eqaul or newer, cloud  update_time 1684226120, local update_time 1684226120
2023-05-20 21:59:46.248457[Thread 0x00007ff84e2e2640]:Found the Preset locally: Polymaker Polylite ABS
2023-05-20 21:59:46.248460[Thread 0x00007ff84e2e2640]:load_user_presetpreset Polymaker Polylite ABS's update_time is eqaul or newer, cloud  update_time 1684311953, local update_time 1684311953
2023-05-20 21:59:46.248474[Thread 0x00007ff84e2e2640]:Found the Preset locally: Silk PLA 0.20mm Standard
2023-05-20 21:59:46.248476[Thread 0x00007ff84e2e2640]:load_user_presetpreset Silk PLA 0.20mm Standard's update_time is eqaul or newer, cloud  update_time 1684042050, local update_time 1684042050
2023-05-20 21:59:46.248494[Thread 0x00007ff84e2e2640]:Found the Preset locally: TailS Calibrated ABS
2023-05-20 21:59:46.248497[Thread 0x00007ff84e2e2640]:load_user_presetpreset TailS Calibrated ABS's update_time is eqaul or newer, cloud  update_time 1684276409, local update_time 1684276409
2023-05-20 21:59:46.248512[Thread 0x00007ff84e2e2640]:Found the Preset locally: TailS Generic ABS
2023-05-20 21:59:46.248515[Thread 0x00007ff84e2e2640]:load_user_presetpreset TailS Generic ABS's update_time is eqaul or newer, cloud  update_time 1684107957, local update_time 1684107957
2023-05-20 21:59:46.248532[Thread 0x00007ff84e2e2640]:Found the Preset locally: TailS Generic TPU
2023-05-20 21:59:46.248535[Thread 0x00007ff84e2e2640]:load_user_presetpreset TailS Generic TPU's update_time is eqaul or newer, cloud  update_time 1684232033, local update_time 1684232033
2023-05-20 21:59:46.248546[Thread 0x00007ff84e2e2640]:Found the Preset locally: Tails PetG
2023-05-20 21:59:46.248549[Thread 0x00007ff84e2e2640]:load_user_presetpreset Tails PetG's update_time is eqaul or newer, cloud  update_time 1684474175, local update_time 1684474175
2023-05-20 21:59:46.248566[Thread 0x00007ff84e2e2640]:Found the Preset locally: Transparent Petg
2023-05-20 21:59:46.248569[Thread 0x00007ff84e2e2640]:load_user_presetpreset Transparent Petg's update_time is eqaul or newer, cloud  update_time 1684474466, local update_time 1684474466
2023-05-20 21:59:46.248583[Thread 0x00007ff84e2e2640]:update_compatibility for all presets enter
2023-05-20 21:59:46.248718[Thread 0x00007ff84e2e2640]:update_compatibility for all presets exit
2023-05-20 21:59:46.249406[Thread 0x00007ff84e2e2640]:save_to_json:1338, saved config to /Users/tailsmedia/Library/Application Support/OrcaSlicer/user/2813097732/process/0.25mm Standard @BBL X1C.json

hamlinjacob, 2023-05-21:
I just had the same issue. Wiped out my filaments, machine settings, and processes which were all saved locally. I do not think there is a way of getting these back.

Everyone should backup their local presets ASAP.

buckeyestargazer, 2023-05-21:
Wow, this happened to me as well.  I am beyond frustrated with this....

TailSMeDiA, 2023-05-21:
what i dont understand is i saved a couple projects with the filaments i created selected and loading them still doesnt make them show up

hamlinjacob, 2023-05-21:
Okay, I think this is an issue with Bambu labs cloud services being down. They should resync the presets when they go back up.

buckeyestargazer, 2023-05-21:
> Okay, I think this is an issue with Bambu labs cloud services being down. They should resync the presets when they go back up.

I sure hope so.  Will setting the printer to "Lan Only" prevent this sort of thing?

hamlinjacob, 2023-05-21:
I have no idea, but I am going to be backing up my presets from here on out to another local folder. 

FYI, the files are stored:
C:\Users\(username)\AppData\Roaming\OrcaSlicer\user\(stringofnumbers)

buckeyestargazer, 2023-05-21:
I just checked the appdata folder and all my presets are gone.  I do have saved .3mf files of many models and I should be able to restore the filament and print settings presets from those (open .3mf and save the changed preset with a new preset).

TailSMeDiA, 2023-05-21:
Just got it all back by syncing!  BACKUP NOW!!!

closed 2023-05-21 by TailSMeDiA