OrcaSlicer/OrcaSlicer #1109: Buildplate appears black

echolot79, 2023-05-21:
**Describe the bug**
-- on the 1.6.3 beta my buildplate appears black 
I´m using a Neptune 3 Pro but the black Plate appears at all Printers in the Settings 
  on 1.6.2 everything was fine 

left 1.6.2  right 1.6.3 beta 

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 pro 64bit 
 - Version 22H2

Landhus, 2023-07-06:
I'm having the same issue as well.

maximoehre, 2023-09-22:
Same issue still on 1.7.0 with Neptune 4 Pro

ReverendRhyme, 2023-10-19:
Same issue on 1.7.0 on Anycubic Vyper and Ender 3 series.

itzner, 2023-10-19:
I could fix the issue by deleting the bed texture file in C:\Program Files\OrcaSlicer\resources\profiles\<manufacturer>, as described in #1187

Ltek, 2023-10-23:
Sam problem on 1.7.0 with Artillery X2... deleting the texture file removed the black but did not replace it with the custom bed texture for the X2

Ltek, 2023-11-11:
1.8 and this is still not fixed?

hjoungjoo, 2024-01-23:
I'm using 1.9.0 and I'm in the same black state on AMD Ryzen 5 3400G.
I expect improvement in the next version.

closed 2024-05-01 by github-actions[bot]