OrcaSlicer/OrcaSlicer #1110: Internal bridge (solid infill) density, flow, fan speed etc.

PhilBaz, 2023-05-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Yes, internal bridging for ABS is critical and requires fine tuning to minimize burt bits and lumps froming in skin... I have tested this much with Cura that has fine control for these settings. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Internal bridge (solid infill top first layer) density, flow, fan speed etc.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Using Cura... but I was instantly impressed with Orca and would love to see this funtionality. 

**Additional context**
Add any other context or screenshots about the feature request here.

Great work! very nice slicer to use....Much better user experience than prusa, superslicer etc.

rendition36, 2023-08-13:
**Is your feature request related to a problem? Please describe.**
The "Internal Bridge Direction" option does nothing, and auto bridge calculations are always weird and sub-optimal.

**Describe the solution you'd like**
To make the function actually work, or maybe smarter internal bridge generation

**Describe alternatives you've considered**
Tried so many different settings and orientations nothing helps.

DanielCard81, 2023-10-06:
That would be very good!

Salmonsquire, 2023-10-06:
Bringing my request over from #2309 (and closing it as duplicate.

> When printing large surfaces, I am having difficulty in dialing in values for the internal bridge layer that don't also impact the external bridges. The result is that there is poorer top surfaces, with occasional pillowing or overextrusion. The current method to resolve this is to lower the bridge width and bridge density. This reduction improves the first solid infill layer.
> 
> This would affect all printers
> 
> The solution I propose is to make it so that internal bridges are able to dialed in independently of default/external bridges. (Possibly even additional control for the interaction between internal bridge and the first solid infill layer)

And @hippotastic 's Post:

> I'd also love to see this feature. I just spent 30 minutes trying to figure out why my internal bridges were printed at 0.4mm layer height although the rest of my print is set to 0.25mm layer height and "thick bridges" are turned off.
> 
> Just like @Salmonsquire reported, the unexpectedly thick internal bridge caused a visible defect in my print as well - a big visible bulge in the inner surface of a bowl. The model was from the "Bauhaus tray" set on Printables.

PhilBaz, 2023-10-06:
@Salmonsquire Awesome thanks! Yeah, let's hope this gets picked up. It's a critical feature.

JeremyWeisener, 2023-10-23:
I've recently been experiencing issues and was looking for a way to control this.
It's happened for me the same as @PhilBaz when using ABS it leaves over extrusion which affects the walls.

More fine grain controls like the ones being proposed would be awesome.

vaselo, 2024-02-29:
Up! please!

ReviloEgros, 2024-03-20:
I'd like to see these options too! Or a way to completely turn off internal bridges!

PhilBaz, 2024-03-21:
@ReviloEgros Ahh, good point having a way to totally disable internal bridges might also be useful in some cases. In Cura this is handled with a sparse infill threshold. Like, no internal bridges with sparse infill > 30%.

rickyT09, 2024-07-06:
Please implement this! This is the only issue that kept me from switching to Orca. So unfortunate because otherwise I like Orca much more than Cura in many regards.

GylRon1, 2024-08-23:
Please add the option to turn off internal bridges because my prints fail at those layers.(it keeps printing, but extrusion is off?)

faberyx, 2024-11-23:
bumo

Hammerfest, 2024-12-25:
putting my thumbs up here as well, the structure I want visually internally can be seen fine from the bottom in test prints, but the top which I inspected in slice is supported properly but still prints a bridge layer :(

madaley1, 2025-02-16:
Bump, having issues with near-glass transparent printing, and I think being able to swap my bridge from aligned rectilinear to rectilinear would solve a good 80% of my issues.

mrambossek, 2025-02-27:
yes please!

PhilBaz, 2025-02-27:
Didn't these just make it into the beta?

Rocheez, 2025-03-14:
This would be great. Having a fan speed high enough for zero-support bridges during an entire infill in low-height models with ABS risks warping and being able to change these settings would help a lot.

heyroman, 2025-04-29:
Bump from me too, glass-transparent printing is messy because internal bridges are different from the rest of the the infill

syao99, 2025-05-07:
This could possibly be solved by simply allowing the "Internal Bridge Density" setting to be set to 0%. Doesn't even need a separate checkbox setting.

PePeGrillo-arg, 2025-05-16:
> This could possibly be solved by simply allowing the "Internal Bridge Density" setting to be set to 0%. Doesn't even need a separate checkbox setting.

you can't set "Internal Bridge Density to 0%, at least 10% yo need to configure

jbjhjm, 2025-08-25:
oh ffs, a bot closing issues randomly after ONE week of silence isn't a good helper, its torpedoing valid feature requests! -_-' With 40+ upvotes it should be clear there's some interest for this.

NairoDorian, 2025-09-14:
BUMPING UP Why can't I disable Internal Bridge ????

hpux735, 2026-03-06:
I just spent like 2 hours trying to figure out how to have an even layer on a debossed color layer.  It keeps looking like crap because the stupid internal bridging

[image]

Why isn't this even!!  blargh!

[image]

here's how it should look