OrcaSlicer/OrcaSlicer #1112: optional backup of all settings offline when sync doesn't work

TailSMeDiA, 2023-05-21:
**Is your feature request related to a problem? Please describe.**
i lost my filament settings when the cloud services went down.  i fortunately got them back when the service went up and i logged in and out 4x.

**Describe the solution you'd like**
we should have an option to do a daily/weekly/monthly backup of all cloud service profiles or all offline files.  not everyone knows where they are savent and an automated process to save filaments/profiles would help when bambu cloud goes offline

**Describe alternatives you've considered**
i can just copy the folder where they are saved to, but not everyone remembers to do that, or some people might not know where they are saved

**Additional context**
just lost all my filament profiles when sync went offline, when it came back i had to log in and out 4x, each time i got 1 or 2 filament presets back.  an automated system would help me not panic when i open and all are gone.

bani6809, 2023-05-22:
bambu labs gave a very poor response to the exact same bug filed on bambu slicer. https://github.com/bambulab/BambuStudio/issues/1840

an option to export **ALL** settings, everything, to a backup file would be nice. that way no matter what happens, you always have a way to restore all your hard work tuning everything.

mattkerrison, 2023-05-26:
> bambu labs gave a very poor response to the exact same bug filed on bambu slicer. [bambulab#1840](https://github.com/bambulab/BambuStudio/issues/1840)
> 
> an option to export **ALL** settings, everything, to a backup file would be nice. that way no matter what happens, you always have a way to restore all your hard work tuning everything.

Agreed, I just opened the configuration folder from the help menu and took a copy incase the cloud goes down again, but either a setting in Orca to periodically backup the presets folder to a directory of our choosing or even a button to do something similar would be a good work around for the moment.

Manually opening the configuration folder and copying the presets out is fine once but will end up with stale presets very quickly.

closed 2023-11-20 by github-actions[bot]