OrcaSlicer/OrcaSlicer #1114: "Detect narrow internal solid infill" results in lots of gaps and weaker parts

EvaFPV, 2023-05-22:
**Describe the bug**
The concentric infill from "Detect narrow internal solid infill" being on leaves lots of gaps in both Arachne and Classic modes resulting in a much weaker part. Somewhat visible in sliced preview, very visible in printed part. 

**3mf File for This Bug**
[DetectNarrowInternalSolidInfill-bug.zip](https://github.com/SoftFever/OrcaSlicer/files/11525928/DetectNarrowInternalSolidInfill-bug.zip)

**To Reproduce**
Slice a model with small areas to be filled with internal solid infill. Compare with setting on and off. 

**Expected behavior**
On or off, solid infill should fill the area leaving no gaps (pinholes maybe but no large gaps)

**Screenshots**
[image]
[image]

**Printer model**
Voron V0.1 with a 0.5mm CHT nozzle

**Desktop (please complete the following information):**

 - OS: Windows 10
 - OrcaSlicer Version 1.6.2

kylek29, 2023-06-06:
I've noticed a similar/same bug, preparing to update some info later as I want to check a few prior OrcaSlicer versions to see where it appeared.

I've tested BambuStudio 1.5 and 1.6 (identical model/settings/filament) and the issue wasn't present.

closed 2023-11-20 by github-actions[bot]

Stormwing01, 2024-07-13:
This is still an issue. The concentric infill seems to cause a lot of holes, which ruins structural integrity, no matter how much you bump up flow ratio.

This setting makes "internal solid infill" in narrow areas under-extrude a LOT, and even increases print time in some cases. the top skin layer is also left under extruded thanks to the holes caused by the under-extruded previous layers.

wackoman6789, 2024-07-28:
Bump this is still an issue and is far worse on longer narrow areas. I've been scratching my head for over a week now because of this from tuning the flow ratio, the temps and the speeds. I just now finally found the setting. 

Storm is right though regular solid infill will look fine and suddenly when it goes to narrow infill it looks like the flow rate was dropped by like 20-30%.
[image]