OrcaSlicer/OrcaSlicer #1117: Temperature Calibration preset for PETG is too high at 260 degrees

klack, 2023-05-22:
**Describe the bug**
Temperature Calibration preset for PETG is too high.

**To Reproduce**
Print a Temperature calibration tower and select PETG.  My PETG started burning and made a really bad smell at 260 degrees.

**Expected behavior**
The default start temp is lower, such as 250.
260 is too high for most common printers which use a PFTE tube in the hotend.  The dangers of this are already well documented.

klack, 2023-05-22:
Right... what I am saying though is the default of 260 is too high.   99% of people are not going to change it.

Naninani, 2023-05-22:
> Right... what I am saying though is the default of 260 is too high. 99% of people are not going to change it. Most people have a pfte hotend.
> 
> I am also struggling to find even one manufacture of PETG that claims a temp of 260.
> 
> The default setting is incorrect, and unsafe.

This is not true. I print Pusament at 270 and it can handle it perfectly well. In addition GreenGate3D can handle up to 290 https://greengate3d.com/collections/recycled-pet-g/products/pink-grapefruit-virgin-pet-g

If your filament is burning it may be the brand that you use.

klack, 2023-05-24:
Wouldn't it be better to manually turn UP the temperature to 260, instead of leaving it at a dangerous temperature for PFTE tubes?

SoftFever, 2023-05-24:
> Wouldn't it be better to manually turn UP the temperature to 260, instead of leaving it at a dangerous temperature for PFTE tubes? This is the point I am trying to get across, not personal preferences or brands of filament.

It's a valid point.
Will change to 250 in next version.

klack, 2023-05-25:
Awesome!  And for the record, I WAS using an exotic filament that burned easily (Matte PETG).  I do have a Capricorn tube which is capable of 260.

closed 2023-08-13 by SoftFever