OrcaSlicer/OrcaSlicer #1118: [Feature Request] Non-planar slicing

tpimh, 2023-05-22:
There are currently several implementations of non-planar slicing. The resulting G-code is using layers that are not flat, effectively using the Z-axis of the printer. The obvious benefits are better surface finishing even if the top of model if not flat and reduced need for support structures, but there might be more. Nozzle geometry should be taken into account.

More information can be found here: https://github.com/supermerill/SuperSlicer/discussions/1741

TexZeTech, 2023-05-23:
I would love to play with Non-planar in a slicer that is actively getting updated. 

*_**making custom hotend intensifies**_*

closed 2023-11-20 by github-actions[bot]

JMVS, 2024-05-07:
Is this of any help? Non-Planar in PrusaSlicer https://github.com/EiNSTeiN-/PrusaSlicer

n3roGit, 2025-05-14:
Reopen :D