OrcaSlicer/OrcaSlicer #1119: Incorrect color order when slicing

bogdi1988, 2023-05-22:
**Describe the bug**
When a model is printed with multiple colors, the slicer chooses a random color for the calibration lines as well as the first filament to be printed.
In the attached screenshot, notice that the model is sliced with blue as the primary color, yet the slicer sets the white accent color as the material that will be used for calibration, as well as the very first section of the first layer that is printed.
The issue is present in BambuStudio 1.6.2 as well. Previous builds of the app would assign the flow calibration to the correct primary object color instead of a random color.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Paint a model with multiple colors. Slice model. Notice the calibration color is not the same as the main model color.

**Expected behavior**
When slicing, the first color to be used is the main, primary color of the model

**Screenshots**
[image]

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

Mrblazed, 2023-05-28:
Please allow us to choose the first color printed on the first layer

foto-boss, 2023-06-11:
I confirm!
I just got stuck with the same problem.

closed 2023-11-20 by github-actions[bot]

danielo515, 2025-06-07:
This is not completed, bot