OrcaSlicer/OrcaSlicer #1125: [feature request] ability to specify the number of layers to combine infill by

DKS-18, 2023-05-23:
Currently it is possible to enable infill combination but not to specify how many layers we want the infill combined by.

When printing at small layer heights like 0.1mm I have found that combining infill can drastically speed up print times. But by default the combined infill prints a thick infill only every 4 layers at 0.4mm thick. This has made for some poor top surface finish on some of my parts.

The above example sliced in superslicer it is possible to specify to print infill every 2nd layer for instance resulting in a 0.2mm thick infill or even every 3rd layer.

I feel that this would be benifical to specify printing every x number of layers. 1 being the default setting where combined infill is disabled

closed 2023-11-20 by github-actions[bot]

R1SA123, 2023-12-10:
strenght > advenced > infill combination

karlc28, 2023-12-28:
can this be reopened as a feature request as its an awsome feature that prusaslicer and cura both have

L3v15, 2024-05-13:
Isn't this more obvious than infill combination?