OrcaSlicer/OrcaSlicer #1127: Move profiles to separate repo

julie777, 2023-05-23:
**Is your feature request related to a problem? Please describe.**

I missed an important notification while watching Orca Slicer because it got lost in the multitude of notifications relating to new printers being added.

**Describe the solution you'd like**

Separating the various profiles into independent repos by type would
1. make selective notification easier because it could be done by repo
2. adding profiles and updating them would not need to be part of an Orca Slicer release. They could be released separately, or users could just download the latest if the wanted.
3. The current repo is grossly oversized with some much information/files from various systems that it is hard to understand.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hliebscher, 2023-05-23:
@julie777  the repo is not oversized and never hard to understand if you develop slicer software. i hope you understand that OrcaSlicer only based on BambuStudio and this is based at Prusa....
all this repos are similar 
the profiles use features and this was add with new versions so we have to add this features to printer / profiles and filaments and please only at one place !

mydevpeeps, 2023-05-31:
Marlin does it this way and I am a contributor to their config repo. I have to say, it's a mixed feeling for me. It can be a pain to have to go off looking at two repos for updates to things. However, it does make life for the contributor a lot easier when the code, and docs, and printer configs, are all separated when it's time to process merge and pull requests. 

The OP could have been a little less abrasive with their comments, but that should not invalidate the topic.

closed 2023-11-20 by github-actions[bot]