OrcaSlicer/OrcaSlicer #1128: Create an Orca Oganization on GitHub to own all of the Orca projects

julie777, 2023-05-23:
If the plan for Orca is to grow beyond being a project that is a personal project of @SoftFever, then it would be nice to create a GitHub organization to own it and any other repos that are added to the project in the future.

This would not actually change anything immediately if @SoftFever created the Organization because he will be the administrator and decide who to add and what roles they have. 

What it would do is signal to the 3D print community that Orca Slicer is more than a pet project and it should be a serious consideration when choosing a slicer. It would allow creation of an Organization page that will be the first thing seen by the community and will provide the overview of Orca that gives users a reason to use it.

I am not going to describe any more detail, but reading the the [documentation ](https://docs.github.com/en/organizations) about organization will show how useful this can be.

hliebscher, 2023-05-23:
@julie777 I love the open-source project by SoftFever, and sometimes, a small developer community is better and easier to organize. Even as it is, it is taken seriously. However, I believe it doesn't always need to be compared to PrusaSlicer, SuperSlicer, and Cura. It is already the best of the slicer worlds for me because it combines all my desired features. But it's not my place to comment on SoftFever's interests. I'm just happy to have contributed to this slicer, not just by reporting issues but also by finding errors myself. Thank you, SoftFever, for your work! We can continue the discussion under "Discussions" if you'd like, as it doesn't really belong here.

julie777, 2023-05-25:
@hliebscher  I was not suggesting that the community should not stay small. I also appreciate the work done by @SoftFever. 

I do think that making Orca a "brand" and adding additional tools that make the ecosystem even better is a reasonable goal. They could of course be created by a different group and be unrelated to Orca Slicer. I just thought that @SoftFever might want to consider the possibility that Orca could be more than just a slicer, and become a full environment, that includes the best slicer.

closed 2023-11-20 by github-actions[bot]