OrcaSlicer/OrcaSlicer #1130: P1P will not connect to camera after turning off LAN only mode

julie777, 2023-05-23:
**Describe the bug**

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
1. I turned off LAN only mode to upgrade firmware.
2. upgraded firmware with Handy
3. Logged in to my BBL account so that Orca could see the printer
4. sent a print to the printer which worked just fine
5. click play on the camera
6. got the dialog about the printer access code and needing to be on the same network (but I am not in LAN only mode.)
7. Closed Orca to see if a normal startup would work
8. started Orca
9. hit play on the camera live feed and received the same error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Printer model**

**Desktop (please complete the following information):**
 - OS: Windows]
 - Version 1.6.2

hliebscher, 2023-05-23:
witch Windows Version? Have you tested with beta 1.6.3? have you restart the Printer and your PC?

julie777, 2023-05-25:
windows 10
not ready to upgrade to beta
yes/yes
It works with Bambu Studio which implies that something is different in Orca. I reported it just to provide notification. I do not need a fix. I have switched back to LAN only mode.

SurgeHawk, 2023-08-10:
I have also encountered this issue. I switched to LAN mode on the  P1S but was unable to send gcode due to another issue [#1720](https://github.com/SoftFever/OrcaSlicer/issues/1720). Turned LAN mode off and now I am not able to view the camera in device tab. Bambu Studio and Handy app work normally.

SurgeHawk, 2023-08-16:
This issue is still present in 1.6.4 beta 3.

closed 2023-11-23 by github-actions[bot]