OrcaSlicer/OrcaSlicer #1131: Retraction calibration

MIG003, 2023-05-23:
On orcaslicer 1.6.3 beta ,  when using  retraction tower calibration in gcode file generated  there is no "SET_RETRACTION " to change retraction length .
I use klipper

hliebscher, 2023-05-23:
witch printer ? or profile ? can you upload your 3mf file as zip that we can reproduce?

MIG003, 2023-05-23:
I have updated my publication whith 3mf file

hliebscher, 2023-05-24:
its the same ... it works not with firmware retraction
at first you have to disable firmware retraction in printer settings.
than print the test 
note the value with no stinging
add the value in your printer start code or firmware retraction settings in klipper

vaninva, 2023-06-10:
> 
Have you solved this problem about UNKNOW COMMAND G10 G11 ？

closed 2023-11-20 by github-actions[bot]