OrcaSlicer/OrcaSlicer #1139: Adding primitive objects to plate has "Orca Cube" with is a little misleading

julie777, 2023-05-25:
## Is your feature request related to a problem? 

I like the idea of having the ability to add primitive objects to the plate. Primitives add the ability to modify a model, add strength, cut out parts etc. They are an alternative to making changes in CAD, and they offer some functionality not possible in CAD. Adding text is a similar function.
 
However, the objects that are being added to the primitives with the addition of Orca Cube and some of the others is both a little confusing and hides another great feature that Orca has added. These are not primitives, but different kinds of calibration models. They deserve to be visible as part of the calibration feature and not hidden away.

Orca Cube is actually multiple objects that have different purposes. Grouping them into one menu item is another way of hiding something that should be visible.

I think that providing calibration objects is a great idea!
This is much easier than tracking them down online. It also allows Orca to choose the best ones. 

## Describe the solution you'd like

I have already submitted an issue to Bambu Studio about hiding the View menu under the File menu. This goes against 30 years of common use and causes users not to discover the different views available, which are very useful. @SoftFever put the calibration menu in the menu bar instead of hiding it. 

I would like to see all the non-file sub-menus moved to the menu bar, so they have equal visibility. I suspect that many Bambu Studio users do not know how useful the different views are, and how much easier it is to switch to the top view for previewing layers is by using the shortcut.

**Adding calibration objects to the calibration menu would make them more visible. It would be obvious what they are used for. It would add value to Orca Slicer.**

Because adding more than a dozen calibration object to the calibration menu, some of them printer specific, would become very cumbersome, I would suggest that a "Calibration Objects" menu item be added, with an indicator showing that a dialog will be opened when it is chosen.

The calibration objects dialog could:
  1. printer model selection to limit models shown.
  2. a list of models, each having a tool-tip and a link to documentation about using it. 

One or more models can be selected and they will be created on a new plate (so they do not interfere with the current project). 

There could also be an option create a new calibration project. The calibration project could include plates with all the calibration models pre-populated. This might be even better. Having a calibration project ship with the slicer, having all the objects that are useful. The calibration project could also be used to run the existing calibration tests. 

One thing that is always missing with calibration/test prints is the lack of information for examining them and what to look for and what it means. Even the temperature tower often lacks this. The links mentioned above addresses this deficiency.

There are some extremely good models out there that users might not find, but if included in Orca could be a big help when diagnosing print issues.

Models that I have used:
  - XYZ cube
  - self created hole sizers that I use for testing press fit of magnets
  - https://www.printables.com/model/395946-the-totally-mundane-lit-calibration-cube is one I really like. It can tell so much about the print in one small model
  - first layer test, such as this https://www.printables.com/model/159909 which is not the one that I used, but the first one I found just now
  - fit together tests such as https://www.thingiverse.com/thing:1601160 and https://www.thingiverse.com/thing:1601177

Everyone prints the benchy, but I never have. I am sure that it can provide much information, but only if you know how to examine it. I do not.

One of the main reasons that users switch to Orca is the calibration tests built in. That might not be the case for non-BBL printer users, but is was certainly the reason for me. This feature request just adds to that part of Orca Slicer.

fr3ak2402, 2023-05-25:
@julie777

So the calibration objects can be placed in a better position, I don't want to deny that. Your suggestion to place them under the menu item calibration is a good one.

I am just a little confused, so with me there is no View item under the File menu item.
So I use the slicer on a macOS system and this is not the case there. 

On which operating system do you use the OrcaSlicer?

---------------------------------------------------
The Benchy can be used to determine many different aspects of the quality of the print, i.e. the printer setting. For example, how well overhangs are printed or how good the flow is.

SoftFever, 2023-05-27:
While these models can be used for calibration purposes to some extent, I personally consider them more as generic test prints rather than specifically designed calibration prints. 
That's why I decided not to include them in the Calibration menu.

I agree `Add Primitive` is the wrong place to put them too.
Shall add a new menu item alongside `Add Primitive`, a menu called `Handy models` or `Test prints` maybe.

fr3ak2402, 2023-05-27:
@SoftFever 
Personally, I also use the objects only for the purpose of a test print. But most of the time they are listed on Thingiverse or Printables as calibration objects, which they actually are not. Many people like you and me only use them for quick test prints to see how the quality of the print is with the settings.

I must also admit that when I installed OrcaSlicer, I first looked in the Calibration tab. I don't think it's a big deal where to find these objects now. 

But yes, maybe we should put this in a separate menu item, as you suggested in `Test prints`.

julie777, 2023-05-29:
@fr3ak2402 
I am using Windows and to me the menu structure is bizarre.
[image]

Thanks for pointing out that I did enter it wrong. In Fusion 360 they buried the view menu under file. In BS/Orca the non-file menus are buried under the arrow icon, but still hidden.

fr3ak2402, 2023-05-29:
@julie777 

I will adapt this in such a way that all menus also stand next to each other in a Windows OS, as in macOS / Linux, and are not grouped together under a fold-out arrow.

So that then is as follows:

File | Edit | View | fold-out arrow with Preferences and Help

Is that all right with you?

julie777, 2023-05-30:
@fr3ak2402 That would be great!

closed 2023-11-20 by github-actions[bot]