OrcaSlicer/OrcaSlicer #1146: add option to define chamber preheat cycle (build plate position + temp, target chamber temp)

ckuhtz, 2023-05-26:
On a printer like the Bambu X1C (or an enclosed P1P), if the build plate is homed and moved near the top, the build plate is set to a user-defined temp (say 120C), the aux fan is turned to high, it can be used to rapidly heat the chamber to the desired temp (user-defined).  Once the target chamber temp has been reached, the normal printing program can then proceed as before.  Essentially, you're convection heating the chamber with the build plate as the aux fan can easily carry away the heat produced by blowing over it.

Build plate temp needs to be driven through the build plate settings as the max will depend on the type (120C for Bambu engineering plate, for example).  Target chamber temp needs to be driven through the filament setting.  With both of these variables, it's ambient independent (although a max cycle time might be a nice safeguard, but not must have). Build plate position for this operation should probably should be tacked on to printer settings; it's desirable to not have it all the way under the nozzle, but 5-10mm below.

(I currently do this manually, but it sure would be nice to automate this step; I have no idea how I'd go about implementing this or I'd already have done it.)

Any takers? Disagree? Agree?

fr3ak2402, 2023-05-26:
@ckuhtz 
The temperature of the printing plate is related to the filament, as each filament has its own physical properties. This means that each PLA is different, for example. You set the temperature by adjusting the first layer or other layers, if you set it too high it will have an effect on the filament. This could lead to an elephant foot.
Therefore, the positioning of the temperature setting has already been correctly chosen by Bambulab. In my opinion, because the temperatures are also related to the filament.

But what we can build into the slicer is that it asks which maximum temperature is permissible for the respective pressure plate.

The field for setting the printing chamber cannot be entered in the filaments, as this does not really make sense. Because the Bambulab has no heating for the chamber and does this via the pressure plate, as you mentioned. You can't set 60°C for the chamber and only 45°C for the pressure plate, how can the printer logically set the 60°C? It is not possible to get more heat than the plate has. Maybe Bambulab will upgrade the hardware for a chamber heater.

You can define in the start code of the printer how fast the aux fan should rotate, and you can also adjust the position of the printing plate. But many functions that happen during printing, such as the fan control, are unfortunately regulated via the printer's firmware, which we unfortunately do not have access to. It could be that you cannot automate all things.

Also, I'm not sure why Bambulab has the aux fan blowing over the print board during heat up, it could be related to board leveling or longevity of the print plate.

We are also not making any changes to the BBL printers, as these are only to be managed by BBL in order to remain compatible with Bambu Studio. If BBL makes a change to the profile, we can more easily keep track of it and build it into OrcaSlicer.

But that doesn't mean you can't change the profile of BBL for you.

In addition, only the BBL printers support multiple plates in the slicer. All third party printers currently support only one printing plate (high temperature plate). This should also change in the future.

sudermanjr, 2023-05-26:
This would be a great feature, and a very common use-case with Voron printers. We're doing this manually right now for printing ASA. @fr3ak2402 I don't believe you fully addressed the core concept of the suggestion by  @ckuhtz with your response.

ckuhtz, 2023-05-26:
@fr3ak2402,

thanks for your reply.  Please _carefully_ re-read what I wrote. 

> The field for setting the printing chamber cannot be entered in the filaments, as this does not really make sense. Because the Bambulab has no heating for the chamber
[..]
> Also, I'm not sure why Bambulab has the aux fan blowing over the print board during heat up, it could be related to board leveling or longevity of the print plate.

It doesn't. 

That's why I suggested the slicer orchestrate the printer in the way I described above: Use the bed heater with airflow to heat the chamber, which is what many of us are doing manually already. By positioning it carefully, you can speed up the process.

Not sure what to say to the rest of your response as it seems completely out of context to me.

Thank you,
Christian

ckuhtz, 2023-05-26:
> This would be a great feature, and a very common use-case with Voron printers. We're doing this manually right now for printing ASA. @fr3ak2402 I don't believe you fully addressed the core concept of the suggestion by @ckuhtz with your response.

Thank you.  Concur, this is something I noted on Bambu X1C but has applicability beyond that as the idea can be implemented anywhere.

There's no reason why OrcaSlicer can't or shouldn't be a superset to whatever Bambu throws out.  Just like Bambu innovated on top of others.

No, manually stitching profiles with custom G-code is neither scalable nor reliable.

fr3ak2402, 2023-05-26:
@ckuhtz 
Sorry I misunderstood you a bit above. I looked at it from the developer's point of view and not from the user's point of view. My mistake, I like your suggestion. I own 5 X1C printers myself. I would build the feature into the slicer and also test it on my X1Cs.

I just have to think about how best to incorporate this into the slicer, since we don't actually want to make any changes to the BBL profiles. Actually this would be the easiest way with the X1C, but in view of the other printers that are suitable for the function, every G-code had to be fitted.

I could write you a G-code for your X1C which will turn on the AUX fan when heating up if you want...
so that you have it automated, you just have to tell me which filaments you want it for.

ckuhtz, 2023-05-27:
Small comment, the fan speed of the aux fan is heighest about 6x  "1-step" toggles down from the top.  Seems that's least resistance for the air to move over the plate. (You can hear it if you home, move plate to top, and then toggle down one step at a time). I don't know what that translates to in terms of dimensions, sorry. Looks like ~10mm below nozzle to me.

ckuhtz, 2023-05-27:
> @ckuhtz I'm sorry that I misunderstood you above, I wouldn't deny that either. I looked at it from the developer's point of view and not from the user's point of view. My mistake, had a long day today.

Alles klar. ;-)
 
> I own 5 X1C printers myself. I would build the feature into the slicer and also test it on my X1Cs.

Cool.  Thank you. Let me know if you'd like help testing it. I'd love to see how this is implemented if you don't mind tagging me on the PR.
 
> I just have to think about how best to incorporate this into the slicer, since we don't actually want to make any changes to the BBL profiles. The reason is that it is more difficult for us to maintain them when Bambulab develops them further, because we then have to adapt a lot etc.

Understand. Does Bambu Studio actually throw up on anything added to the json that it doesn't recognize or is it just ignoring it? If it's ignoring the property, then we can just add an object and enumerate properties within that as a superset.
 
> But no matter, we'll get through that with a suggestion to install it in the slicer.

Thanks!
 
> Actually this would be the easiest way with the X1C, but in view of the other printers that are suitable for the function, every G-code had to be fitted.

Right. I think there needs to be an option in the machine profile to enable per machine.  Once enabled (and specified with the appropriate G-Code), the options in the filament and process become live.  ILR, this could become a feature to turn on chamber heaters for machines that have it instead of using the bed heater + fan + position of the build plate as a workaround.
 
> I could write you a G-code for your X1C which will turn on the AUX fan when heating up if you want... so that you have it automated, you just have to tell me which filaments you want it for. Or rather, does this feature make sense for all filament types?

I personally only care about things like ASA.  But PC and others could be a candidates. I don't know what an X1C does if you were to get to 90-100C for chamber and 180C for bed. :-) So, ABS, ASA, and other filament/process ilk that are in that same temp range would be targets on a first go around.

For PLA, this doesn't make much sense obviously.

> Should the feature be switched on and off? Then I can also add an option on the UI.

Yes. I think it needs to be controllable and something to enable explicitly for a process profile. In the Others tab maybe?  Call it "[ ] Chamber pre-heating" with a tooltip like "Enable this to pre-heat the chamber temperature to the desired target temperature.  Requires support in machine profile."  Ideally, it'd be greyed out if the machine profile doesn't have the magic bits to tell it "yes, this machine can do it" and "this is the g-code for how to do it".

So, correction to the first post, filament doesn't need to know about this.  But the machine and process profiles do (and associated UI).  

Happy to help any way I can.

SoftFever, 2023-05-27:
I did similar things on my Voron.
Take a look at print_start/HEATSOAK macros I wrote here: https://github.com/SoftFever/Voron2.4_350_SoftFever/blob/master/macros.cfg 

But I don't see a chance to achieve that with Bambu printers.
Bambu users are at the mercy of Bambu when it comes to achieving a high level of freedom and customization.
I won't hold my breath ;)

SoftFever, 2023-11-06:
There's nothing the slicer can do in this case except issue a command instructing the printer to wait until a certain temperature or time is reached, as this involves reading sensors and controlling hardware. The printer's firmware must support these commands. While this is straightforward with open-source firmware, it's worth noting that BambuLab's printer firmware is not open-sourced.

closed 2023-11-06 by SoftFever

markaudacity, 2024-06-15:
This issue is definitely not completed as of 1.9.1