OrcaSlicer/OrcaSlicer #1149: Print Plate Hotkey (CTRL+SHIFT+G) opens different dialog than Print button and does not function

triangleman83, 2023-05-27:
**Describe the bug**
The hotkey designated as "Print Plate" which is CTRL+SHIFT+G opens a dialog which appears to be a Bambu cloud print dialog. This does not provide an option to connect to the host printer. This dialog seems to be from Bambu Slicer and does not function properly to send the print to a non-Bambu printer. See screenshot below:

[image]

**To Reproduce**
Load any STL and change to Preview tab. Wait for slicing to complete. Press CTRL+SHIFT+G on the keyboard to open the above dialog. No printer option is shown for a local LAN hosted Ender 3. The Send button does not function. The link opens a page at Bambu's website.

**Expected behavior**
The dialog box which is loaded should show the host printer as an option with the Send button uploading the gcode to the host. Additional information may be shown on this screen such as the filename of the gcode to be uploaded. The dialog which opens when pressing the Print button at the top right of the preview tab shows a filename for the gcode and uploads to the host properly. See screenshot below of that dialog.

[image]

**Printer model**
Ender 3v2 with Raspberry Pi 3A running klipper and fluidd.

**Desktop (please complete the following information):**
 - OS: Windows 10 
 - Version 10.0.19045 Build 19045
 - OrcaSlicer 1.6.2

SoftFever, 2023-05-27:
@triangleman83 
Thanks for reporting the issue.
This issue has been fixed in 1.6.3-beta

triangleman83, 2023-05-27:
Works great for me, thank you!

closed 2023-05-27 by triangleman83