OrcaSlicer/OrcaSlicer #1150: Ruler

rkarlsba, 2023-05-27:
When importing a model, sometimes it could be very nice to have a ruler of sorts to measure the distance between two points. Could this be added?

liftbag, 2023-05-27:
I vote for porting the measure tool from PrusaSlicer, where it is present starting from version 2.6.0-alpha2 and which has been further improved in apha6.

SoftFever, 2023-05-28:
Gonna port PS 2.6 measurement tool in the future.
That should cover this use case I believe.

ZerkerEOD, 2023-07-29:
Has this been added yet?

cpasoft, 2023-08-21:
> Has this been added yet?

Not yet. Hopefully soon. 💪🙏

GTreaper, 2023-10-06:
Big + on this

EyalRotem, 2023-10-31:
@SoftFever  - is there any idea when will this be implemented?

SoftFever, 2023-10-31:
> @SoftFever - is there any idea when will this be implemented?

I’ve put a PullRequest with the initial port.
But we changed the plan after discussing with @Noisyfox 
He will port the whole related PrusaSlicer framework so that we can port most new PS tools in one go as both of us think Prusa’s tool(cut and text) is better

EyalRotem, 2023-10-31:
> I’ve put a PullRequest with the initial port. But we changed the plan after discussing with @Noisyfox He will port the whole related PrusaSlicer framework so that we can port most new PS tools in [...]

That’s great news. So it is planned for say.. the next minor release? 1.9.0?
Or possibly through 1.8.x updates?

cpasoft, 2023-10-31:
> 
> But we changed the plan after discussing with @Noisyfox He will port the whole related PrusaSlicer framework so that we can port most new PS tools in one go as both of us think Prusa’s tool(cut and text) is better

I love that... Of course Prusa's cutting and text tools are better...

SoftFever, 2023-10-31:
Most likely 1.9

Noisyfox, 2023-10-31:
Here are some progress:

[image]

[image]

closed 2023-11-19 by SoftFever

xros, 2023-11-26:
> Here are some progress:
> 
> ![image](https://private-user-images.githubusercontent.com/1537155/279423641-c266e33e-e5ac-4fb5-8764-73a02d28bc6b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5NTkwMDAsIm5iZiI6MTcwMDk1ODcwMCwicGF0aCI6Ii8xNTM3MTU1LzI3OTQyMzY0MS1jMjY2ZTMzZS1lNWFjLTRmYjUtODc2NC03M2EwMmQyOGJjNmIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjZUMDAzMTQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmVjMTJiYmZlOTllOWFhMmY4NjlmOTNjMDY4MDVmMzUzNTcxZmE5NTkwZDAyN2RhZTI3Y2RhNmVlOTU3ODc2MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.77nzVagGcKBkA1T9reSeRixwb4tyYrebI5uDEtWWEEk)
> 
> ![image](https://private-user-images.githubusercontent.com/1537155/279423508-decced7f-7b87-4282-b97f-aa9d2d4d2122.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5NTkwMDAsIm5iZiI6MTcwMDk1ODcwMCwicGF0aCI6Ii8xNTM3MTU1LzI3OTQyMzUwOC1kZWNjZWQ3Zi03Yjg3LTQyODItYjk3Zi1hYTlkMmQ0ZDIxMjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjZUMDAzMTQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGUzNmE5MWEwMzMyYzQ1YzdlOTlmZTVkZDVhMmY2MDI4NDg5YzQyYjk5NWRhZjQ2YjRlMjgzNGRjYmExNzI5YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.rhIdC53f1Ev2guvMMKSbADMPQBBENVsPa-B-AI8TegI)
> 
> No ETAs but I'll do as quickly as I can.

Please add the rule! This is very useful!
 Thanks. It looks so good!

cpasoft, 2023-11-26:
> Please add the rule! This is very useful! Thanks. It looks so good!

It is already added. It will be avaliable on next release, or you just download the last nightly.

xros, 2023-11-26:
> It is already added. It will be avaliable on next release, or you just download the last nightly.

Thanks I will download the nightly build now. :) wonderful

Noisyfox, 2023-12-04:
It will be in 1.9

ajaytanna, 2023-12-04:
Thank you very much.

samstone86, 2023-12-16:
Is it available from nightly build ?

Noisyfox, 2023-12-16:
> Is it available from nightly build ?

yes

stef1949, 2024-01-06:
> > Is it available from nightly build ?
> 
> yes

YIPEEEE :3

PR #2603 (2023-11-19): Port PrusaSlicer's Measure & Cut gizmos
Build on top of #2520 and replaces it.

This PR includes:
- Rewrite the opengl rendering code, which now renders (almost) everything using shaders instead of legacy opengl function calls
- Rewriting the gizmo mouse handling code that moves the mouse handling coding into each gizmo themselves
- Rewriting the mouse picking code, now it uses ray casting to figure out what's under the mouse cursor
- Porting of the PrusaSlicer's measure tool
- Replacing existing cuting tool with the better one PrusaSlicer has
- Updating of other gizmos using PrusaSlicer's lat [... cut]