OrcaSlicer/OrcaSlicer #1152: Option to do outer wall first, except for when there's an overhang?

mathewsamueldesign, 2023-05-27:
I am trying to get better outer wall quality. Many people say printing the outer wall first is better for quality, but there's the issue with overhangs and printing in thin air. Is it possible to add an option where it prints outer walls first, except for when there's an overhang? When there's an overhang, it's probably better to print inner first.

liftbag, 2023-05-27:
It could be a good idea, even if I fear that a different finish between the two areas could be highlighted.

SoftFever, 2023-05-28:
My immediate thought is:
We will have to split the outer perimeters many times to perform such switch operations.
That will result in creating many many extra z seams @_@

closed 2023-11-20 by github-actions[bot]