OrcaSlicer/OrcaSlicer #1154: Aligned seams aren't aligned... and why is there no 'corner priority' option?

nirurin, 2023-05-27:
As you can see, this is set to be aligned, but the seams are all over the place. 
If I change to 'back' it then aligns the seam into a single corner... which makes me wonder why aligned doesn't do the same thing? 
Do some options prioritise corners, while others don't? 
If so, why isn't there an "aligned-corner" option, when that's the option that people use 99% of the time? 

[image]
 
[Paladins-Io-Bow V2Arm_Skeleton.zip](https://github.com/SoftFever/OrcaSlicer/files/11583126/Paladins-Io-Bow.V2Arm_Skeleton.zip)

SoftFever, 2023-05-28:
`Aligned` has corner priority by default.
In your case, you are using negative parts to cut the part, which confused the seam placement algo(it's potentially a bug still)

I suggest using the cut tool as a workaround:
[image]

nirurin, 2023-05-28:
Ahh ok I see. I thought the cut tool could only do horizontal cuts not vertical ones. I'll give it a try

closed 2023-11-20 by github-actions[bot]