OrcaSlicer/OrcaSlicer #1155: Layer level in Preview is double layer level when printing

ChuckVanzant, 2023-05-28:
When comparing the layers completed in the Device tab as the print is in progress to the associated layer in the Preview tab, the layer count is double in the Preview tab. As you step through each layer in the Preview tab, you can see the layer count increments by 2. 

Is this the result of using the 0.28mm process? Is this the expected behavior?

OrcaSlicer Version: V1.6.2
Sliced an object using **0.28mm Extra Draft @BBL X1C**
A zipped .3mf file is included.

[Poop Chute Remix v3.zip](https://github.com/SoftFever/OrcaSlicer/files/11583551/Poop.Chute.Remix.v3.zip)

SoftFever, 2023-05-28:
I can't repro the issue with the latest version(1.6.3)

I'm going to release the 1.6.3 official release later today. Please give it try later

ChuckVanzant, 2023-05-28:
I downloaded OrcaSlicer_V1.6.3-beta_Win64 and the problem is indeed gone.
Thanks!

closed 2023-05-28 by ChuckVanzant