OrcaSlicer/OrcaSlicer #1159: Portugues Translation

manuelchaves, 2023-05-28:
hello, I would like to help translate Orca into Portuguese BR, how do I proceed?

SoftFever, 2023-05-28:
@manuelchaves 
Cool.
I will see if I can generate a empty Portuguese .po file for you to work with.

manuelchaves, 2023-05-28:
Ok, do I need some specific app or direct txt?  As a reference I have boards from Lerdge, INaV and others.  Free of charge, just as a thank you to the excellent Orca.  tks

SoftFever, 2023-05-28:
@manuelchaves 
I've create a branch for Portuguese (Brazilian) language. [
feature/pt_BR](https://github.com/SoftFever/OrcaSlicer/tree/feature/pt_BR)

The file to work with is https://github.com/SoftFever/OrcaSlicer/blob/feature/pt_BR/bbl/i18n/pt_BR/OrcaSlicer_pt_BR.po
I usually use a text editor to modify .po files, but I can imagine that translating a file from scratch would be a bit cumbersome.
 
@hliebscher, you seem to have more knowledge on this topic. Could you provide some advice on the available tools for editing .po files?

manuelchaves, 2023-05-28:
I really like POEdit https://poedit.net/download simple but with excellent features that help a more cohesive translation.  It also accepts xliff and json files

SoftFever, 2023-05-28:
I should give POEdit a try next time

SoftFever, 2023-05-28:
Moving it to discusstion/dev forum