OrcaSlicer/OrcaSlicer #1160: Macro parameter of `chamber_temperature` for different tool?

thisiscam, 2023-05-28:
**Is your feature request related to a problem? Please describe.**

I found that orcaslicer currently only uses the `chamber_temperature` from the first tool (first filament in the filaments tab), even when the first filament might not be used in the print. 

**Describe the solution you'd like**

Ideally, we should be able to access the `chamber_temperature[extruders]` parameter same as in [superslicer]( https://github.com/supermerill/SuperSlicer/wiki/Macro-&-Variable-list), so that I can issue something like `START_PRINT T_CHAMBER=[chamber_temperature[initial_tool]]`. But, it seems like this "array variable" is not implemented in orcaslicer, and I am not even sure if orcaslicer parses nested expansions at all.

**Describe alternatives you've considered**
We could also pass `chamber_temperature` as the temperature of the first tool, that would solve the problem in most cases.

SoftFever, 2023-05-28:
Makes sense

SoftFever, 2023-08-13:
Will be available in 1.6.4 beta 3

closed 2023-08-13 by SoftFever

FlorianWilk, 2023-08-13:
Is this fix including the fix for the Fan-Speed (and maybe other Filament-values), or should i open up a seperate issue for that?